labatrockwell/openTSPS #85: 1.3.7 crashes on startup - Mac OS X 10.10.3

cylinderStudio, 2015-05-01:
Process:               TSPS [519]
Path:                  /Users/USER/Downloads/*/TSPS.app/Contents/MacOS/TSPS
Identifier:            cc.tsps.tsps
Version:               1.0
Code Type:             X86 (Native)
Parent Process:        ??? [1]
Responsible:           TSPS [519]
User ID:               501

Date/Time:             2015-04-30 21:51:40.155 -0700
OS Version:            Mac OS X 10.10.3 (14D136)
Report Version:        11
Anonymous UUID:        DDAFB4B2-2864-CFAA-A460-0ACE5A2F52BD

Time Awake Since Boot: 580 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libPS1080.dylib               	0x0cc174cc usb_get_next_device + 577
1   libPS1080.dylib               	0x0cc11b10 darwin_get_device_list + 63
2   libPS1080.dylib               	0x0cc0e43f libusb_get_device_list + 82
3   libPS1080.dylib               	0x0cc0c7a5 xnUSBEnumerateDevices + 53
4   libPS1080.dylib               	0x0cbcaad3 XnDeviceEnumeration::Initialize() + 275
5   libPS1080.dylib               	0x0cbbfe4b XnOniDriver::initialize(void (*)(OniDeviceInfo const*, void*), void (*)(OniDeviceInfo const*, void*), void (*)(OniDeviceInfo const*, int, void*), void*) + 219
6   libPS1080.dylib               	0x0cbbda36 oniDriverInitialize + 54
7   libOpenNI2.dylib              	0x00eabf2e oni::implementation::DeviceDriver::initialize() + 62
8   libOpenNI2.dylib              	0x00ea7185 oni::implementation::Context::loadLibraries(char const*) + 1077
9   libOpenNI2.dylib              	0x00ea6d15 oni::implementation::Context::initialize() + 1189
10  libOpenNI2.dylib              	0x00eb71c1 oniInitialize + 33
11  cc.tsps.tsps                  	0x0004b52a ofxNI2::init() + 266
12  cc.tsps.tsps                  	0x0004bc1f ofxNI2::Device::setup() + 15
13  cc.tsps.tsps                  	0x00065076 ofxTSPS::GuiManager::setup() + 6646
14  cc.tsps.tsps                  	0x00026091 ofxTSPS::PeopleTracker::setup(int, int, std::string, int) + 1585
15  cc.tsps.tsps                  	0x00023230 ofxTSPS::Delegate::setup() + 96
16  cc.tsps.tsps                  	0x00009967 tspsApp::setup() + 343
17  cc.tsps.tsps                  	0x005646fd Poco::PriorityDelegate<ofBaseApp, ofEventArgs, false>::notify(void const*, ofEventArgs&) + 61
18  cc.tsps.tsps                  	0x0004d2bc Poco::AbstractEvent<ofEventArgs, Poco::PriorityStrategy<ofEventArgs, Poco::AbstractPriorityDelegate<ofEventArgs> >, Poco::AbstractPriorityDelegate<ofEventArgs>, Poco::FastMutex>::notify(void const*, ofEventArgs&) + 220
19  cc.tsps.tsps                  	0x005699bc ofNotifySetup() + 44
20  cc.tsps.tsps                  	0x005d707a ofAppGLFWWindow::runAppViaInfiniteLoop(ofBaseApp*) + 42
21  cc.tsps.tsps                  	0x00560f9c ofRunApp(ofBaseApp*) + 1260
22  cc.tsps.tsps                  	0x00008ee2 main + 178
23  cc.tsps.tsps                  	0x000029b5 start + 53

Thread 1:
0   libsystem_kernel.dylib        	0x986a4e6a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x95d0c2b1 _pthread_wqthread + 939
2   libsystem_pthread.dylib       	0x95d09e2e start_wqthread + 30

Thread 2:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x986a58ce kevent64 + 10
1   libdispatch.dylib             	0x9908d73f _dispatch_mgr_invoke + 245
2   libdispatch.dylib             	0x9908d3a2 _dispatch_mgr_thread + 52

Thread 3:
0   libsystem_kernel.dylib        	0x986a4e6a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x95d0c2b1 _pthread_wqthread + 939
2   libsystem_pthread.dylib       	0x95d09e2e start_wqthread + 30

Thread 4:
0   libsystem_kernel.dylib        	0x986a4e6a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x95d0c2b1 _pthread_wqthread + 939
2   libsystem_pthread.dylib       	0x95d09e2e start_wqthread + 30

Thread 5:
0   libsystem_kernel.dylib        	0x986a4e6a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x95d0c2b1 _pthread_wqthread + 939
[... 345 lines cut ...]
CoreUI image data                    32K
IOKit                              8572K
IOKit (reserved)                      4K        reserved VM address space (unallocated)
Kernel Alloc Once                     4K
MALLOC                             76.3M
MALLOC (admin)                       48K
MALLOC_LARGE (reserved)            1500K        reserved VM address space (unallocated)
Memory Tag 242                       12K
Memory Tag 249                      156K
OpenCL                               16K
Stack                              68.6M
VM_ALLOCATE                        17.2M
__DATA                             23.8M
__GLSLBUILTINS                     2588K
__IMAGE                             528K
__IMPORT                              4K
__LINKEDIT                         47.3M
__OBJC                             2648K
__TEXT                            181.0M
__UNICODE                           552K
mapped file                       121.3M
shared memory                         4K
===========                      =======
TOTAL                             556.3M
TOTAL, minus reserved VM space    554.8M

robotconscience, 2015-05-01:
Hm! Is this from the downloaded binary? I'm on the same version + OS X and can't replicate.

What device(s) do you have plugged in, and what are you trying to use? We'll figure this out!

cylinderStudio, 2015-05-01:
Yep, it's from the download at Sourceforge. Tried 1.3.7 and 1.3.6. I don't have any devices plugged in. Initially I had an XBox 360 Kinect #1473, but I unplugged it and the external Apple display and restarted my MacBook Pro and tried again, same result.

robotconscience, 2015-05-01:
Super weird. Do you have a webcam on your machine? Any camera spoofing apps such as CamTwist?

Double check the settings in bin/data/settings. Make sure defaults.xml has a device index of 0. Just did a clean download and haven't had issues, so weird! Your crash is from the OpenNI2 library, so not 100% sure how to fix that yet until I can replicate it.

cylinderStudio, 2015-05-01:
Just the built-in iSight camera on the MacBook. No spoofing apps or anything like that. I think the only camera apps I have at all are Skype and Photo Booth.

Confirmed defaults.xml has a device index of 0.

I was actually getting these crashes before I installed OpenNI2. But I've tried since uninstalling that and I guess I wasn't successful. I'll try again to make sure that's wiped and download another 1.3.7 of TSPS and try again.

robotconscience, 2015-05-01:
OK! It comes with the OpenNI2 libs in the binary, so that shouldn't matter. Anything else you think might be unusual in the way you're launching it?

cylinderStudio, 2015-05-01:
Hmm...not really. Just downloading it, unzipping it and running it directly from the freshly unarchived folder by double-clicking.

I just manually uninstalled OpenNI2 from the binary I had been using (sudo ./install.sh -u) in case there was some conflict, and got a new 1.3.7 zip that downloaded and unzipped right away. Same result. Head-scratcher. I'm wracking my brain for anything else that might have installed something that conflicts. Leap Motion? Um...

robotconscience, 2015-05-01:
Head scratcher indeed. You could try removing the OpenNI2 drivers and see if that works:
* Right click "Show package contents" on TSPS
* Delete everything in MacOS/Drivers
* Launch again!

cylinderStudio, 2015-05-01:
Interesting. I had installed libfreenect (and libusb) with homebrew. Anyway, I left my Kinect plugged in and just deleted everything in the package under MacOS/Drivers and relaunched. It didn't crash. The window opens (black, no UI) and I get a beach ball. Doesn't crash after a full minute. Did a force-quit (as it's not responding) and unplugged Kinect, tried again and TSPS runs. Works normally with my built-in iSight.

robotconscience, 2015-05-01:
Oof, very weird. OK, try restoring those drivers one at a time! It looks like OpenNI is trying to grab that Kinect, which makes sense. I need to phase that lib out in the end.

cylinderStudio, 2015-05-01:
Progress! If I include only libOniFile.dylib alone, it works. I can access the Kinect or the iSight, and switch back and forth. Including libFreenectDriver.dylib on its own causes the beach ball scenario. Including libPS1080.dylib on its own causes the crash.

robotconscience, 2015-05-01:
Really good to know. I'll be looking at all this as I get Kinect 2 into everything, so will keep that in mind! Hope that is at least a good workaround for you for the time being.

cylinderStudio, 2015-05-01:
Yep, totally works for me for now. Thanks for helping me out.

cylinderStudio, 2015-05-04:
Would one of the temporarily removed drivers affect Websocket communication at all? When I open the debug URL/localhost:port, I'm getting an error.

robotconscience, 2015-05-05:
Nope! Definitely log as a separate issue. Sorry about that; most likely just an error in the JS.

cylinderStudio, 2015-05-05:
No sweat. Logging now.