labatrockwell/openTSPS #86: Error with debug URL for socket server (1.3.7)

cylinderStudio, 2015-05-05:
When I open the debug URL/localhost:port, I'm getting this error in the browser:

Connected browser: ...
Not initialized

main.js:49
"Cannot read property 'x' of undefined"

robotconscience, 2015-05-05:
Just tested, totally just a tiny bug in the code.

If you send the contours from TSPS, you won't see the error. The fix is to replace the block that starts at line 49 with this:

```
if ( person.contours.length > 0 ){
				ctx.beginPath();
				ctx.moveTo(person.contours[0].x*vid_width,person.contours[0].y*vid_height);

				for (var j=1; j<person.contours.length; j++ ){
					ctx.lineTo( person.contours[j].x*vid_width,person.contours[j].y*vid_height );
				}				
				ctx.stroke();
			}
```

robotconscience, 2015-05-05:
Sorry, this is all in bin/data/web/js/main.js!

cylinderStudio, 2015-05-06:
Thanks! That definitely helped, but in Tracking, if I set maximum blob size to anything over 10% of view, the socket server quits with:

WebSocket connection to 'ws://localhost:7681/' failed: Invalid frame header

robotconscience, 2015-05-06:
Ah! Try turning contours off for now and using that fixed javascript if you can.

I know what the error is, need to update the Websockets library and recompile. There was a bug in that addon that's now fixed. Will post here when it's updated!

cylinderStudio, 2015-05-08:
Works for me. Thanks again.

robotconscience, 2016-02-11:
Fixed in feature-kinect2 branch (soon to become master branch)

commit e2e702ce (2016-02-11): Closes #86
closed 2017-11-19 by robotconscience