labatrockwell/openTSPS #89: syphon input isn't working?

jeffro-, 2015-08-11:
Hey I have a Syphon server open that for some reason doesn't show up in TSPS. I click the Syphon option and it just shows a black screen. Syphon works just fine in other apps. Anyone know what the problem is? I'm on OS X 10.10 if that helps.

ajilderda, 2016-06-06:
I'm experiencing the same problem when trying to send out Processing frames through a Syphon server into openTSPS. Did you find a solution?

ajilderda, 2016-06-06:
Also the Syphon output from OpenFrameworks unfortunately does not work. Is this a known problem? @robotconscience

robotconscience, 2016-06-06:
Hey @ajilderda, I'm not sure! In general is ofxSyphon not working? Totally weird–I may have to update some underlying frameworks. Apologies for the delay, have not been able to focus on TSPS for a bit on my side.

ajilderda, 2016-06-06:
Thanks for your reply, @robotconscience . ofxSyphon is working fine. I ran the examples and the output is received in the Syphon demo app (Simple Client), but not showing up in TSPS. This is also the case with the output from Processing. This also shows up in Simple Client, but not in TSPS.

But... When the Syphon demo Simple Server app then it IS received by TSPS. Hmm... Am I doing something wrong?

I dug into this as far as my knowledge reaches. Hope you find the time to help out, as this could be of incredible use for a project I'm working on!

robotconscience, 2016-06-06:
Can you clarify which stuff _is_ working?

Is TSPS to Syphon demo working, or Syphon demo to TSPS working?

As best as I can tell (I wrote the code a long time ago! :) ), the Syphon app (input) name is "Simple Server". Does that help? The Syphon server name (the output) is "tspsSyphon", so if you bind to that it should work on the output side, I think.

robotconscience, 2016-06-06:
OK had a chance to test on my side... The input into TSPS I'm not sure of! The input into processing appears to be sort-of working, but it's drawing a mesh as outlines for me. Not sure what's up there–are you having luck in general with the processing library?

ajilderda, 2016-06-06:
Sorry for being unclear. :) One more time. I was refering to 'Simple Client' and 'Simple Server', the test apps that are on the [Syphon](http://syphon.v002.info/) website ([which you can download here](https://github.com/Syphon/Simple/releases/download/public-beta-2/Syphon.Demo.Apps.Public.Beta.2.dmg)).

✅ OpenFrameworks > Syphon Client
✅ Processing > Syphon Client
🚫 OpenFrameworks > TSPS (black screen)
🚫 Processing > TSPS (black screen)
✅ Syphon Server > TSPS (the demo video is correctly shown in TSPS, hmm...)

Indeed, I didn't had the output server name set to 'tspsSyphon'. I just changed it in the Processing and OpenFrameworks script but still get the black screen. 

Is it necessary for the app to be named "Simple Server" in order to cast video to TSPS correctly (which I tried for the ofx build, but same problem)? How would that work for Processing?

Thanks!!

ajilderda, 2016-06-06:
As far as I can tell the _input_ into processing is working quite well, except for the fact that I'm also seeing a mesh as outlines in the Syphon example.

ajilderda, 2016-06-07:
It was indeed the Syphon app input name that needs to be "Simple Server". After changing the 'Product name' in build settings to 'Simple Server' the video gets picked up by TSPS. 👍 Not sure how this would work for Processing though (as you have no control over the app name).

Thanks for your help!

robotconscience, 2016-06-07:
So weird! I bet the product name sets the "app name" or "server name" in syphon... I'm sure you can play with both of those in P5 and get something that works. Sorry TSPS seems to be tied to that placeholder name, will add a text field for it in the future!

ajilderda, 2016-06-08:
Especially for TSPS multicam that would be a great feature (so that you're able to use different Syphon inputs).