labatrockwell/openTSPS #92: Multicam project out of date?

mattfelsen, 2016-03-01:
I cracked open the multicam Xcode project and it looks like it's a pre-0.9 file. Just wondering if you had to do anything weird/special to get the basic Xcode project up to date, or if the standard PG + adding the necessary addons/libs/frameworks to mirror the basic project flow should do it?

robotconscience, 2016-03-01:
I think that should be it! It's really much easier now with the addons that aren't in the local "addons" folder. Let me know if you have issues.

I think as we discussed the other day, it'd be awesome to start re-thinking this app. Right now it's just 4x instances of TSPS (or less if you use command-line args). Would be slick to try to do some simple stitching...

Promise I'll try to dig up the one we did for Google...

mattfelsen, 2016-03-01:
Yea that'd be the dream...configurable for 2x2 vs 1x4 grid, etc. I'm also into trying to merge point clouds and do tracking on that instead of stitching depth maps. All in due time...

For my project I think we're using Kinect2, which means one cam per computer (I don't trust multiple cams on one machine being stable), which means multiple computers, so no rush on digging it up :smile:

mattfelsen, 2016-03-02:
Guessing the Windows side needs the same update, yea? Same question: anything funky to be aware of with conflicting Kinect1/Kinect2 libusb stuff, that preprocessor def for Kinect2, winsock/winsock2 stuff? I'll start with whatever's in the readme but not sure if it's out of date! Also hoping at this point it's possible to have a single sln/vcxproj for both Kinect1/Kinect but not sure if that's doable

robotconscience, 2016-03-02:
RE: last point, I think it's doable! The main thing is the drivers, totally don't work at the same time. Otherwise, not bad. You'll most likely have to remove libusb from Kinect 1, that's about it!

mattfelsen, 2016-03-13:
I spent an embarrassingly and frustratingly long time trying to get the latest version to compile on Win today without any luck :tired_face: I kept some rough notes as I went, which I'm including below for any future attempts (by me or otherwise :stuck_out_tongue_closed_eyes:)

For now, I'm just going to use the older 0.8.4/Win32 version from last summer for this project, and if I add any new stuff (gonna try to do depth map -> point cloud w/ ortho camera to compensate for lens distortion/people bowing out away from the center), I can try to get it into the latest version later :+1: 

- removed win_cb section from ofxLws and NI2 addons
- addons.make: removed osx-specific stuff (syphon, openglcontextscope, kinectv2)
- addons.make: added local addons (..\..\addons\ofxLabGui, opencv extensions, ofxtsps)
- remove ofApp.h/cpp files that projectGenerator includes
- remove doxyfile from ofxtuiowrapper
- add ofxkinectforwindows2 via rbarazza's instructions...mostly...his 2012 version is slightly out of date for vs2015
- TSPSDelegate.h (addons/ofxTSPS/extras) not found, needed to add header path (PG error?)
- winsock.h -> winsock2.h as per Brett's notes
- manually add ofxOpenNI include files to project & header paths since PG didn't pick up either?
- TuioClient::connect issue regarding char/wchar for creating mutexes. replace "cursorMutex" with L"cursorMutex" etc
- ofxNI2 XnPlatform.h instead of #error'ing for MSVC > 1600, add #include "Win32/OniPlatformWin32.h"
- added `#define TSPS_KINECT2` somewhere
- eventually pulled out ofxOpenNI, ofxNI, OpenNIProcessor, etc.

- added this inside PeopleTracker.h `namespace ofxTSPS` a al 0.8.4 version
```
#ifdef TSPS_KINECT2
	// a little helper
	typedef Kinect2 Kinect;
#endif
```

¯\_(ツ)_/¯

mattfelsen, 2016-03-13:
Also should add this was OF 0.9.3, with the projectGenerator built from source on master (https://github.com/openframeworks/projectGenerator/commit/da225803e3d1c6408ada33d18e22b2e9fb611c0d), and I think all addons are on their current master.

robotconscience, 2016-03-13:
Oof, sorry buddy.

Can you post some of the errors you're getting here? So many funky things it could be; lots of weird things with ofxLws and some addons / windows things (this is a common thing: http://stackoverflow.com/questions/1372480/c-redefinition-header-files-winsock2-h/1372836#1372836). 

Will help as best as I can, only have one active Win machine rn but it's at work!

mattfelsen, 2016-03-14:
Right, sorry, I didn't even include the last errors where I left off! The last thing issue I saw before giving up for the day was "multiply defined symbols found" relating to ZLIB.lib. The PG had set up ZLIB.lib and zlib.lib as linker inputs (I think the PG is being funky currently), but removing one of them didn't help. I also tried pulling out everything related to LWS, Spacebrew, the lws & zlib libs, etc. but then I got unresolved linker errors, something about opencv looking for some compression functions that are probably in zlib.

Thankfully nothing related to winsock/winsock2 or libusb conflicts (yet?) (aside from the stuff you put in the readme already which seemed to have done the trick)!

I'll get a better error log/screenshot when I boot back into Windows sometime this week :expressionless: